roboticslab-uc3m/teo-developer-manual #2: Yarpmotorgui control modes

rsantos88, 2017-02-27:
Implementar todas las funciones necesarias para que Yarpmotorgui quede totalmente funcional.

rsantos88, 2017-02-28:
Fixed error `[ERROR] iint->getInteractionlMode failed` implementing IInteractionMode.cpp with empty functions (https://github.com/rsantos88/teo-body/commit/420d59dfa623e9c64ca2bbc99a1b71f91e8b8689).
To do: complete implementation!

rsantos88, 2017-02-28:
Implement functions are missing... [here!](https://github.com/rsantos88/teo-body/blob/develop/libraries/BodyYarp/TechnosoftIpos/ITorqueControlRawImpl.cpp)

rsantos88, 2017-03-07:
After adding the update of the new [libraries](https://github.com/rsantos88/teo-body/tree/issue%2390/libraries/BodyYarp/TechnosoftIpos), when I try to change to Velocity mode, it shows this warning message:
```bash 
[warning] ICanBusSharerImpl.cpp:366 interpretMessage(): 	-Mode "0" not specified in manual, may be in Fault or not enabled yet. canId(9).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 180).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 280).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 180).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 280).
[warning] ICanBusSharerImpl.cpp:366 interpretMessage(): 	-Mode "0" not specified in manual, may be in Fault or not enabled yet. canId(9).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 180).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 280).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 180).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 280).
[warning] ICanBusSharerImpl.cpp:366 interpretMessage(): 	-Mode "0" not specified in manual, may be in Fault or not enabled yet. canId(9).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 180).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 280).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 180).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 280).
[warning] ICanBusSharerImpl.cpp:366 interpretMessage(): 	-Mode "0" not specified in manual, may be in Fault or not enabled yet. canId(9).
```
The interface change to blue color but I can't move the velocity bar...

[image]

PeterBowman, 2017-03-07:
Yarpmotorgui calls `IVelocityControl2::setRefAcceleration` [here](https://github.com/robotology/yarp/blob/a0af248bfc2b454772b617aa18ff85e17cd50245/src/yarpmotorgui/partitem.cpp#L947), but I don't see this method implemented in your code: [a](https://github.com/rsantos88/teo-body/blob/issue%2390/libraries/BodyYarp/CanBusControlboard/IVelocityControl2.cpp), [b](https://github.com/rsantos88/teo-body/blob/issue%2390/libraries/BodyYarp/TechnosoftIpos/IVelocityControl2RawImpl.cpp) + http://www.yarp.it/classyarp_1_1dev_1_1IVelocityControl2.html + http://www.yarp.it/classyarp_1_1dev_1_1IVelocityControl2Raw.html.

rsantos88, 2017-03-07:
Trying to change to torque mode, the interface doesn't turn to other color and it shows this error in launchLocomotion terminal:
```bash
[warning] ICanBusSharerImpl.cpp:366 interpretMessage(): 	-Mode "251" not specified in manual, may be in Fault or not enabled yet. canId(6).
```

rsantos88, 2017-03-07:
Fixed torque mode issue with this [commit](https://github.com/rsantos88/teo-body/commit/cf81ab7305451ae3114af9528e173c2599e56bab)

rsantos88, 2017-03-07:
Returning with the problem to change to velocity mode, I've checked the status of the joints and I can see that the velocity mode is changed  successfully:
```bash
yarp rpc /teo/rightLeg/rpc:i
get encs
Response: [is] encs (67.469238 0.0 0.087873 0.0 0.790861 -0.333923) [tsta] 1425 1488901790.066102 [ok]
get icmd cmds
Response: [is] [cmds] ([pos] [pos] [pos] [vel] [pos] [pos]) [tsta] 2440 1488901823.83269 [ok]
```

jgvictores, 2017-03-07:
Mini-comment: this type of issues belong to https://github.com/roboticslab-uc3m/yarp-devices :smirk:

jgvictores, 2017-03-08:
This issue was moved to roboticslab-uc3m/yarp-devices#103

closed 2017-03-08 by jgvictores