roboticslab-uc3m/teo-developer-manual #3: Teo-Main PC out-of-date

jgvictores, 2017-03-07:
Hola,

Es posible que el Pc-teomain del laboratorio esté desactualizado y que algunas cosas no funcionen?? y en caso afirmativo, para cuando podría estar listo?? gracias

rsantos88, 2017-03-07:
Me imagino que te refieres a los nuevos repositorios que habeis cambiado. Mañana a primera hora me pongo con ello. Actualizo también los PCs de Teo.
Un saludo

jgvictores, 2017-03-07:
@rsantos88 Jejeje esto fue @jmgarciah con mi cuenta....

@jmgarciah Sí, hemos cambiado mucho software, hemos roto cosas pero todo es para mejor... Ahora hasta tenemos un repositorio de visión que se llama **vision**, uno de cinemática y dinámica llamado **kinematics-dynamics**... :smile:  Pronto estará todo funcional de nuevo... ¡estamos contigo!

PeterBowman, 2017-03-07:
Ya que ha surgido este tema, apunto también por aquí que quedan por arreglar algunas referencias en la wiki, que a su vez redundan en un cambio de configuración en los PCs que usáis y que por ese mismo motivo dejé como estaban: http://robots.uc3m.es/index.php/User:BartekPL/migraci%C3%B3n_repos. El comentario de `scripts/gnome/***.desktop` se refiere a que en [scripts/gnome/teo-main.desktop](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/845178b/scripts/gnome/teo-main.desktop) y similares hay rutas que no basta con cambiar en el código, también habrá que renombrar directorios allá donde este se use. Y por último, falta por redirigir las páginas de documentación generadas en robots.uc3m.es.

jgvictores, 2017-03-07:
@PeterBowman Síiii... aún nos queda mucho por hacer...

De hecho el nuevo https://github.com/roboticslab-uc3m/teo-main está totalmente en fase YCM-playground :sweat_smile:

jgvictores, 2017-03-09:
This issue was moved to roboticslab-uc3m/teo-main#2

closed 2017-03-09 by jgvictores