roboticslab-uc3m/teo-developer-manual #10: Import from wiki

jgvictores, 2017-04-04:
Import from wiki:
- [ ] http://robots.uc3m.es/index.php/TEO_Diagrams
- [ ] etc, etc, etc...

jgvictores, 2017-12-07:
Blocking https://github.com/roboticslab-uc3m/teo-main/issues/15

rsantos88, 2017-12-12:
mmm... this is [the result](https://github.com/roboticslab-uc3m/teo-software-manual/blob/issue%2310/appendix/a-teo-diagrams.md) using `pandoc`, following the workflow...
I think... it's necessary to do it manually..

rsantos88, 2017-12-13:
done [here](https://github.com/roboticslab-uc3m/teo-software-manual/pull/20)

jgvictores, 2017-12-13:
Great!
Just update the originals (http://robots.uc3m.es/index.php/TEO_Diagrams etc) and feel free to close!

rsantos88, 2017-12-13:
updated!

closed 2017-12-13 by rsantos88

jgvictores, 2017-12-13:
http://robots.uc3m.es/index.php/Procedimiento_Demos ? :smile:

jgvictores, 2017-12-13:
Updated http://robots.uc3m.es/index.php/Procedimiento_Demos