roboticslab-uc3m/teo-developer-manual #14: restore important wiki information

rsantos88, 2017-11-21:
There have been changes in the wiki where important information has been deleted:
http://robots.uc3m.es/index.php?title=TEO&diff=next&oldid=2750

Important to restore:
> CuiAbsolute: VirtualBox Image with Windows 7 and MPLAB IDE v8.92, Documentation.

rsantos88, 2018-02-15:
Do you want to export this page to a `.md` file and include it in this repository?

jgvictores, 2018-02-15:
That would be great!

rsantos88, 2018-02-16:
done at (ceef3b31c60198a5159eddaaffa1120ddfd8b665)

jgvictores, 2018-02-16:
Cool, essentially what had been lost was CuiAbsolute:
- [https://drive.google.com/a/uc3m.es/folderview?id=0BxR76I90oKSmdnRhQlpsS3pXWm8&usp=sharing VirtualBox Image with Windows 7 and MPLAB IDE v8.92]
- [[CuiAbsolute Documentation|Documentation]]. -> imported at ceef3b31c60198a5159eddaaffa1120ddfd8b665.

jgvictores, 2018-02-16:
> [https://drive.google.com/a/uc3m.es/folderview?id=0BxR76I90oKSmdnRhQlpsS3pXWm8&usp=sharing VirtualBox Image with Windows 7 and MPLAB IDE v8.92]

Incorporated info at 1ffe43001115ffff6c05a73727a5fee02c8da3dd

> [[CuiAbsolute Documentation|Documentation]]. -> imported at ceef3b3.

Linked from manipulation and locomotion at same commit, 1ffe43001115ffff6c05a73727a5fee02c8da3dd

jgvictores, 2018-02-16:
Merged into `master` at ee838c71dc2d7befe18889dc531db3fd660d3e5f.

@rsantos88 Thanks a lot! Feel free to close this issue!!

closed 2018-02-19 by rsantos88