roboticslab-uc3m/teo-developer-manual #15: Dynamic information is wrong

AlvaroMartinezR, 2017-11-24:
I realized the dynamic information about TEO from SolidWorks is not completely accurate:

- The center of mass is not related to the coordinate axis of each piece in some cases, but to the global CoC.

- SolidWorks works with different CoC other than the rest of the models we have of TEO (seen in the next image, bottom left corner). For example, here Y is the vertical axis whereas in the rest the vertical is Z. Therefore all information in  teo-software-manual/assets/teo-dynamic-information.md has its axis changed. 
[image]

I just wanted to make it known in case someone wanted to use them. In the meantime,  teo-simox-models/doc/CoM simox.csv can be used, but I am pending for the next week to use ods2md in this file.

jgvictores, 2017-11-25:
I've been looking at [teo-simox-models/doc/CoM simox.csv](https://github.com/roboticslab-uc3m/teo-simox-models/blob/aab15379813f1f69af9d526cfb6f7b0d056d42cd/doc/CoM%20simox.csv) and can say that `.csv` is a perfectly acceptable format to be used directly (instead of `.md`, and no passing through `.ods` or therefore `ods2md`). Essentially what we are looking for is:
- Easy for a human to read.
- Easy for a machine to parse.
- Nice GitHub preview is a plus.

This criteria is met for `.csv` (even have the nice preview on the Desktop version of GitHub).

AlvaroMartinezR, 2017-11-28:
Now is corrected and better presented [here](https://github.com/roboticslab-uc3m/teo-software-manual/blob/master/assets/teo-dynamic-information.md). 

I close the issue.

closed 2017-11-28 by AlvaroMartinezR