roboticslab-uc3m/teo-developer-manual #17: Import http://robots.uc3m.es/index.php/TEO_Network_information

jgvictores, 2017-12-04:
- Import http://robots.uc3m.es/index.php/TEO_Network_information
- Update links to:
  - https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-jr3.md
  - https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-netgear-a6100.md

Help: Workflow described [here](https://github.com/asrob-uc3m/tutoriales/issues/2#issue-273074070)

rsantos88, 2017-12-12:
> - Import http://robots.uc3m.es/index.php/TEO_Network_information

done (c76f305dbe5b0ad2ad716e4011c803836787c0a1)
>- Update links to:
>  - https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-jr3.md
>  - https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-netgear-a6100.md

where do you want to add this links exactly?

jgvictores, 2017-12-12:
Looking at c76f305, there are sections dedicated to installing JR3 (and possibly netgear) that are duplicates. These sections should be removed and replaced by the links you mention.

rsantos88, 2017-12-12:
perfect, that's what I wanted to know

rsantos88, 2017-12-12:
As you've said me before, feel free to modify whatever you want that `pandoc` has not converted correctly. I've appended the links like a  `Other hardware configuration`
Modify it as you want (#18)

jgvictores, 2017-12-12:
Yeah, it's great. Already merged and have enhanced it quite a bit already: [state at time of writing](https://github.com/roboticslab-uc3m/teo-software-manual/blob/136a50a855bbac821b5bbb434a83fc52f314ac0e/overview/teo-network-information.md)

Following the workflow described [here](https://github.com/asrob-uc3m/tutoriales/issues/2#issue-273074070), all that is left is to update http://robots.uc3m.es/index.php/TEO_Network_information and close.

jgvictores, 2017-12-12:
All updated at 5523acc (ended up moving it to appendix)
http://robots.uc3m.es/index.php/TEO_Network_information also updated.

Closing! @rsantos88 Thanks!

closed 2017-12-12 by jgvictores