roboticslab-uc3m/teo-developer-manual #30: Add It2 info on this repo from info given on roboticslab-uc3m/QA#6

jgvictores, 2018-02-02:
_From @jgvictores on February 25, 2017 10:55_

Add It2 info on this repo from info given on https://github.com/roboticslab-uc3m/QA/issues/6 

_Copied from original issue: roboticslab-uc3m/teo-hardware-manual#8_

jgvictores, 2018-02-02:
_From @smcdiaz on February 25, 2017 11:52_

Do you mean the values for each joint? Or the procedure?

jgvictores, 2018-02-02:
Hi!

I put this issue more like a reminder to myself that there's some valuable insight in https://github.com/roboticslab-uc3m/QA/issues/6 and it should be given visibility in this repo.

Anyway, all contributions are always welcome! It would be great to have both the current values for each joint, as well as the procedures here!

PeterBowman, 2021-05-01:
I2t protection values are stored in https://github.com/roboticslab-uc3m/teo-ipos-backup, see https://github.com/roboticslab-uc3m/teo-hardware-issues/issues/45#issuecomment-759550140.