roboticslab-uc3m/teo-developer-manual #40: Inconsistencies in the DH drawing

PeterBowman, 2018-11-21:
The orientation of the TCP is not correct, compare [rightArmKinematics.ini](https://github.com/roboticslab-uc3m/teo-configuration-files/blob/8ced4c99e4137a48aff87397c274943b0aef32cb/share/kinematics/rightArmKinematics.ini) (same for the left arm) and [dh-drawing.png](https://github.com/roboticslab-uc3m/teo-developer-manual/blob/dcab8e5983bccc29f77f5802ade3be3d5080b0a5/assets/dh-drawing.png). That picture was imported from an older version of http://robots.uc3m.es/index.php/File:DH.png and it doesn't reflect the actual behavior of the manipulators (regardless of whether the grippers have been rotated or not).

Also, the naming of some axes is wrong (check *x4* and *x5* on the right arm) or missing (left arm).

jgvictores, 2018-11-21:
I'd consider this a bit of a duplicate of https://github.com/roboticslab-uc3m/teo-main/issues/38

Note that it points to https://github.com/roboticslab-uc3m/teo-main/issues/23 (where I mention to please follow the [workflow-for-assets](https://github.com/roboticslab-uc3m/teo-developer-manual/tree/e88693d314a04309cfbacc09b9bb6636b176e4b4#workflow-for-assets))

PeterBowman, 2018-11-21:
Please either update kinematics-related .inis at teo-configuration-files or fix the diagram here, this is a bug that shouldn't have anything to do with the physical changes described at https://github.com/roboticslab-uc3m/teo-main/issues/23. I feel that https://github.com/roboticslab-uc3m/teo-main/issues/38 had been opened as a tracking issue for double-checking things, whereas I want to stress here that a nontrivial problem was found and it should not be merely taken as implied in a (to some extent) unrelated repo for which I didn't even know such an issue existed.

Could we just replace the diagram with its latest revision from the wiki?

jgvictores, 2019-04-08:
Using this issue to track https://github.com/roboticslab-uc3m/teo-developer-manual updates now that we have closed the grand "Review and Unify Joint Directions of Rotation (https://github.com/roboticslab-uc3m/teo-main/issues/39)"!

jgvictores, 2019-04-08:
- Trunk: no changes

jgvictores, 2019-04-08:
See https://github.com/roboticslab-uc3m/teo-main/issues/38#issuecomment-480930979 for use of https://github.com/arcoslab/roboview (on Trunk for now).

jgvictores, 2019-04-10:
- Right Arm: some changes
[image]

jgvictores, 2019-04-11:
- Left Arm: same changes as for right arm, not copying images again

jgvictores, 2019-04-11:
- Right Leg and Left Leg: some changes, both very similar except for sign of `l16` (`D` of link 4), note how the lower links hav a slight inwards tilt
[image]
[image]

jgvictores, 2019-04-11:
- Head: no changes
[image]
[image]

jgvictores, 2019-04-11:
- Transformations: adding new root to right/left arms, will also update old transformation from hip to arms
[image]
[image]

jgvictores, 2019-04-11:
Regrading https://github.com/roboticslab-uc3m/teo-developer-manual/commit/6ed9975cd4279ac5b0dafbc861ef71d0ada10f26, you can even stack them up, say `dh-root-rightArm.csv`+`dh-rightArm.csv` and `dh-root-leftArm.csv`+`dh-leftArm.csv` to make sure everything is correct from `root` to end-effector

[image]

jgvictores, 2019-04-11:
- Transformations: adding new root to right/left legs, will also update old transformation from root to legs
[image]

jgvictores, 2019-04-11:
Regrading https://github.com/roboticslab-uc3m/teo-developer-manual/commit/b7cf504f8b21c9d0eeb3b0b69afccad0a0e062a5, again we can concatenate, say `dh-root-rightLeg.csv`+`dh-rightLeg.csv` and `dh-root-leftLeg.csv`+`dh-leftLeg.csv` to go "Live on two Legs":

[image]

jgvictores, 2019-04-11:
Due to what is illustrated in the pic, we are changing the head joint order (tracked at https://github.com/roboticslab-uc3m/teo-main/issues/48), to use the DH below:

[image]

jgvictores, 2019-04-12:
> Due to what is illustrated in the pic, we are changing the head joint order (tracked at [roboticslab-uc3m/teo-main#48](https://github.com/roboticslab-uc3m/teo-main/issues/48)), to use the DH below:

False alarm, we can use a solution that involves the neck starting in a lower position as in the figure (closing [roboticslab-uc3m/teo-main#48](https://github.com/roboticslab-uc3m/teo-main/issues/48) and https://github.com/roboticslab-uc3m/teo-configuration-files/pull/19 as `invalid`):

[image]

jgvictores, 2019-04-12:
- Head updated at https://github.com/roboticslab-uc3m/teo-developer-manual/commit/ad023dd74d1c24f9507005458a91620886c31e72 and https://github.com/roboticslab-uc3m/teo-developer-manual/commit/d5f15384e30a66fa5038a5f3313df03dcebaec91 is now taller because starts below:
[image]
[image]

jgvictores, 2019-04-12:
Regarding these commits, again we can concatenate, `dh-root-head.csv`+`dh-head.csv` for the final trunk+head adopted solution:
[image]
[image]

jgvictores, 2019-04-12:
Big merge at https://github.com/roboticslab-uc3m/teo-developer-manual/commit/0c8c519c48edc89e06cc2212419661ad909f1bb7

- [ ] Pending: render the new `csv/dh-root-*.csv` files in the gitbook.

jgvictores, 2019-04-22:
> * [x]  Pending: render the new `csv/dh-root-*.csv` files in the gitbook.

Added at https://github.com/roboticslab-uc3m/teo-developer-manual/commit/13d36106fd6a9c45e986e7ba203520ee8c71151a

jgvictores, 2019-04-22:
As a next step of this task, I'd focus on the end-effectors (kinematic chain tips):
- [ ] `rightArm`/`leftArm`:
   - Reduce `l_8`, most probably to include the F/T sensor but not any hand
   - Add transformations from this end-effector to TCPs for different types of hands
- [ ] `rightLeg`/`leftLeg`: Maybe we'd want some additional changes for walking. Maybe we do not even want root to foot but the opposite. Must put some thought on this.
- [x] `head`: Transformations seem relatively okay for now, but may be subject to change in the future (depends on `vision`).

rsantos88, 2019-04-24:
tested [trunkAndRightArmKinematics](https://github.com/roboticslab-uc3m/teo-configuration-files/commit/08407fa1ccdf29cb75d5a277d4bc373d125fd3f4) and [trunkAndLeftArmKinematics](https://github.com/roboticslab-uc3m/teo-configuration-files/commit/3d584e14e52f0e837c51ba79aad9c51d8a12c589) on teoSim and it works great!

jgvictores, 2019-04-25:
In addition to the [comments above](https://github.com/roboticslab-uc3m/teo-developer-manual/issues/40#issuecomment-485465734), will also focus on scripts to generate `H0` and `HN` transformations. Note that `teo-configuration-files` say they use `asibot-main/example/cpp/testTinyMath.cpp`: https://github.com/roboticslab-uc3m/teo-configuration-files/blob/23703e1e29182a43c491efe04d4f5a9427d6008a/share/kinematics/leftArmKinematics.ini#L3

However:
- `asibot-main/example/cpp/testTinyMath.cpp` is now [roboticslab-uc3m/kinematics-dynamics/examples/cpp/exampleYarpTinyMath/exampleYarpTinyMath.cpp](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/530561b45c238a6f18cffa30d20d90bb25ac587c/examples/cpp/exampleYarpTinyMath/exampleYarpTinyMath.cpp)
- I'd prefer to use `PyKDL`

jgvictores, 2019-04-29:
> However:
>
> * `asibot-main/example/cpp/testTinyMath.cpp` is now [...]

Opened https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/177 for this task.

jgvictores, 2019-04-30:
> `rightArm`/`leftArm`:
> * Reduce `l_8`, most probably to include the F/T sensor but not any hand
> * Add transformations from this end-effector to TCPs for different types of hands

Done via https://github.com/roboticslab-uc3m/teo-developer-manual/commit/4a2431a7804bacf2e0b378101a4a208c768e8012.

Pending: update diagrams.

jgvictores, 2019-05-06:
Done at https://github.com/roboticslab-uc3m/teo-developer-manual/commit/cabf02d5438e2735a4983b82f9ff0c6fca2fe6cd

closed 2019-05-06 by jgvictores

jgvictores, 2019-05-06:
Attaching DH coord systems for Fetch
[image]