roboticslab-uc3m/teo-developer-manual #41: Track and document state of PC that used to be called teo-main

jgvictores, 2018-11-27:
Track and document state of PC that used to be called `teo-main`.

As of https://github.com/roboticslab-uc3m/teo-developer-manual/blob/dcab8e5983bccc29f77f5802ade3be3d5080b0a5/appendix/c-teo-network-information.md#teo-main this info is outdated.

PS: I'm pretty sure I'm the one who did it, but have to go back to the lab to see what happened.

rsantos88, 2019-04-29:
As far as I know, this computer is used to work, like any other computer, connected to the TEO network by ethernet. Does it make sense to keep calling it teo-main? Should we rename it with another name?

jgvictores, 2019-04-29:
https://github.com/roboticslab-uc3m/teo-developer-manual/blob/a83379613c4d8ccd6183375fd6916412521c205e/network-information.md is now free from `teo-main`.

I'm closing this issue as `invalid`, please open any required new issues on https://github.com/roboticslab-uc3m/laboratorios regarding organization of PCs!!

closed 2022-05-24 by PeterBowman