roboticslab-uc3m/teo-developer-manual #46: Export  TEO Network Information to labs repo

jgvictores, 2019-02-09:
Export  TEO Network Information to labs repo: https://github.com/roboticslab-uc3m/teo-developer-manual/blob/a727cda044a9796951f0a63aa72ce84325b65a6c/appendix/c-teo-network-information.md

Essentially only manipulation, locomotion and head make sense here.

rsantos88, 2019-04-29:
Updated only with TEO information: https://github.com/roboticslab-uc3m/teo-developer-manual/commit/836798abc02943efbbaec7197296e29ca01c0892 
Note: yarp-server configuration is now located on manipulation computer section.
Other computer information exported to: https://github.com/roboticslab-uc3m/laboratorios/blob/develop/teo/computers.md

jgvictores, 2019-04-29:
Small reminder to ourselves: https://github.com/roboticslab-uc3m/teo-developer-manual/commit/836798abc02943efbbaec7197296e29ca01c0892#diff-768a7b1f3e2828c1656f22d87b3253c3R112 has already updated `head` ubuntu version. There is actually an issue for that, but no problem IMHO.

Done via https://github.com/roboticslab-uc3m/teo-developer-manual/commit/836798abc02943efbbaec7197296e29ca01c0892 merged at https://github.com/roboticslab-uc3m/teo-developer-manual/commit/a83379613c4d8ccd6183375fd6916412521c205e [ thanks @rsantos88 !!! ]

jgvictores, 2019-04-29:
PS: much cleaner http://robots.uc3m.es/gitbook-teo-developer-manual/network-information.html

closed 2019-04-29 by jgvictores

jgvictores, 2019-04-29:
> There is actually an issue for that, but no problem IMHO.

This one: https://github.com/roboticslab-uc3m/teo-main/issues/49