roboticslab-uc3m/teo-developer-manual #48: Update switches picture and procedure to turn on

jgvictores, 2019-03-19:
Update switches picture and procedure to turn on

rsantos88, 2019-04-26:
Changes: https://github.com/roboticslab-uc3m/teo-developer-manual/pull/57
Any way you can render/visualize the changes in this branch to see it?

PeterBowman, 2019-04-26:
>  Any way you can render/visualize the changes in this branch to see it?

In https://github.com/roboticslab-uc3m/teo-developer-manual/pull/57/files, click on the "..." and view file.

rsantos88, 2019-04-26:
mmm... is not that. I mean to visualize the change in the rendering of the gitbook

rsantos88, 2019-04-26:
I understand that the only way you can render the changes is to download the repository, position yourself in the branch and render it locally.

jgvictores, 2019-04-28:
Closed by @rsantos88 via https://github.com/roboticslab-uc3m/teo-developer-manual/pull/57 merged at https://github.com/roboticslab-uc3m/teo-developer-manual/commit/1d46801c15b018298ce2fcd5a16f58fefc7a28ce THANKS!!!!

jgvictores, 2019-04-28:
PS: Link to rendered page: http://robots.uc3m.es/gitbook-teo-developer-manual/switches-and-power.html

closed 2019-04-28 by jgvictores