roboticslab-uc3m/teo-developer-manual #63: Update motor information in BOM

jgvictores, 2019-07-13:
Update motor information in [BOM (Bill Of Materials)](https://github.com/roboticslab-uc3m/teo-developer-manual/blob/master/csv/bom.csv). Ref: https://github.com/roboticslab-uc3m/teo-configuration-files/issues/23#issuecomment-510988224

rsantos88, 2019-11-08:
Maybe do you want to add an URL in **Desciption/Comments** linking to another table in a separated file with [this](https://github.com/roboticslab-uc3m/teo-configuration-files/issues/23#issuecomment-510988224) information?

PeterBowman, 2019-11-08:
I'm currently working on this via https://github.com/roboticslab-uc3m/yarp-devices/issues/237.

BTW: note that current motor information in BOM is ***WRONG***. We've had incorrect torque constant ("k") values in TEO's arms since quite a long ago.

jgvictores, 2019-11-09:
Marking as blocked by https://github.com/roboticslab-uc3m/yarp-devices/issues/237

PeterBowman, 2019-11-15:
I'd be very grateful for the exact Maxon motor references in CAN ids: 6, 7, 13, 14, 15, 17, 19, 21, 23, 25, 27, 28. Everything else has been described at https://github.com/roboticslab-uc3m/teo-configuration-files/commit/d7b87bbaaa5eb22a384665e051398e594915816a.

cc @smcdiaz @jmgarciah

PeterBowman, 2019-11-19:
Had spoken with @smcdiaz. We use 70 W EC-flat Maxon motors everywhere except for both wrists and the two neck joints, which hold a [339282](https://www.maxongroup.com/maxon/view/product/339282) 30 W motor. The 70 W one can be either [402686](https://www.maxongroup.com/maxon/view/product/402686) or [411815](https://www.maxongroup.com/maxon/view/product/411815), the only difference between those two is the wiring. Also, I noticed that some of the motors that can be visually inspected do not have any reference number on their casing. @smcdiaz suggested those are previous models of the same 402686/411815, although you can easily notice that external dimensions don't match.

That being said, I'd assume the following:

* maxon-402686: ARH6, ALH7, FT14
* maxon-411815: AT13, FRS15, ARS17, ARW19, FLS21, ALS23, ALW25

I'd still review the motor information downloaded into the iPOS drives and configurable via EasySetup.

PeterBowman, 2019-11-20:
Following the suggestion of @smcdiaz, I checked several motors against their EasySetup configuration regarding detected motor+driver resistance and inductance values, thus inferring the 30/50/70 W motor power (yes, we have 50 W motors) by simply comparing with known data: [motors.zip](https://github.com/roboticslab-uc3m/teo-developer-manual/files/3870413/motors.zip) (includes both arms, head and joints 7, 8 and 13).

PeterBowman, 2020-01-28:
Motor ref numbers updated in .ods file (https://github.com/roboticslab-uc3m/teo-developer-manual/commit/d6e4b0e9ad8b23b34c16b3d6043b28a93bc2132d) and ported to .csv (https://github.com/roboticslab-uc3m/teo-developer-manual/commit/c29041b3bb61acdf033b0c86b108d70b970f0c47).

PeterBowman, 2020-01-28:
Done at https://github.com/roboticslab-uc3m/teo-developer-manual/commit/9f944f3f4414fcbcdd5375eab31b0779ee6731f6.

closed 2020-01-28 by PeterBowman